platisd/smartcar_shield #1: Change from MPU-3050 to MPU-6050

DouglasPearless, 2016-01-12:
Would you be able to add support for the MPU-6050 board ( GY-521 gyroscope module ) in addition to the existing MPU-3050 (GY-50 gyroscope module)?

Cheers
Douglas

DouglasPearless, 2016-01-12:
Hmm, re-reading the documentation is that it appears that L3G4200D is the supported chip not MPU-3050, any way, would you be able to add MPU-6050 support?

platisd, 2016-01-12:
Hi, I have an MPU-6050 module lying somewhere around and it's been a while since I used it (2.5 years to be exact!).

Are you interested in just software support or hardware as well, on the shield?

If it's the former, then I might be able to do something, as I want to steadily decouple the library from the hardware shield. Currently I am adding support for steering with a servo and throttling using an ESC. This is expected to be done in the next two weeks. Afterwards, I could add the MPU-6050 since it's a fairly popular component. If you have any piece of code that works well with it, you can let me know so I can use it as a reference and hopefully finish the new feature faster.

However if you are talking about hardware support on the shield, I cannot promise that I will do it, as it is a big change and I need to be convinced first that it's better than the existing approach. Of course, you are always welcome to create a fork of the shield that has slots for MPU-6050! :)

DouglasPearless, 2016-01-12:
Hi Dimitris,

Just the software support, please.

I have seen some Arduino code for it and I will see if I can find it again.

Servo steering and ESC will be interesting, but for the moment I am using two DC motors with gearboxes to drive my robot.

Will need to add a compass too!

Cheers
Douglas

platisd, 2016-01-12:
Cool, will do after I am done with the other stuff that are currently being done!

Regarding the compass, the problem with the (cheap) compasses, according to my experience, is that they pick a lot of interference, becoming unstable and therefore they will not necessarily work out of the box. This will ruin the user experience, possibly leading novice users to frustration. :P
Therefore I don't think it makes sense to add a Compass class for example in the library, for the time being.

DouglasPearless, 2016-01-12:
Here is a link to a good (IMHO) tutorial on using the MPU-6050 http://diyhacking.com/arduino-mpu-6050-imu-sensor-tutorial/ <http://diyhacking.com/arduino-mpu-6050-imu-sensor-tutorial/>

platisd, 2016-01-12:
I'll give it a look when the time comes and will let you know when the new feature will be added, thanks!

DouglasPearless, 2016-01-12:
Thanks, I look forward to trying the code :-)

platisd, 2016-01-29:
Hello again @DouglasPearless ,

I present you with the [Smartcar MPU6050](https://github.com/platisd/smartcar_MPU6050) library. It is an extension of the Smartcar shield library, for the MPU6050, which is essentially a stripped down version of the [MPU6050 Arduino library](https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050), by Jeff Rowberg. I did not include it in this, "core" library, because it is rather... extensive. So, if someone wants to use it, will have to include that specific header file.

Furthermore, it can be used by the smartcar, the same way as the Gyroscope, namely, it can be utilized to make the car [rotate](../wiki/API-documentation#void-rotateint-degrees). You can also `getAngularDisplacement()`, as you did with the [Gyroscope](../wiki/API-documentation#int-getangulardisplacement).

You will find further documentation and examples in the Smartcar MPU6050 library's [readme](https://github.com/platisd/smartcar_MPU6050/blob/master/README.md).

There are some known issues, however aside of that it should be working very well most of the time. If you have problems with it, open an issue in the other library. :)

closed 2016-01-29 by platisd

DouglasPearless, 2016-01-30:
Great, I will have a look at it in a while.