platisd/smartcar_shield #3: getSpeed and GetDirection

DouglasPearless, 2016-01-30:
As there are DCMotors::setSpeed and DCMotors::setAngle methods, would it be possible to have DCMotors::getSpeed and DCMotors::getAngle methods as well to enable the code to determine where the robot is heading?

Cheers
Douglas

platisd, 2016-01-30:
Hi,

They actually have those methods, which are implemented in the parent classes. The DCMotors class extends the SteeringMotor class as well as the ThrottleMotor class (since in some cases you use them for both steering and throttling).

Therefore, you can getAngle and getSpeed from them. However, I would advise against using the DCMotors class directly in your sketches. There is a reason there is no example using any kind of Motor class directly. :P
This is because the Car class abstracts everything motor related, so you can use the same code, regardless of the motors you have on your vehicle. (E.g. you could be using a servo for steering and a motor connected with an ESC for throttling or a servo for steering and DC motor for throttling or even just DC motors for steering and throttling). The code should be essentially the same across all those settings. That is why the Car class is there, so the user deals with it and not with the different motors on each occasion. :)

To sum up, if the user wants to getSpeed or getAngle, he should talk to the Car and not the motors, as the motors could vary depending on your setting. Please refer to the [API documentation] (https://github.com/platisd/smartcar_shield/wiki/API-documentation) regarding the usage of the various library components.

Last but not least, getSpeed and getAngle in any case, will NOT give you the "actual speed" of the car or the "actual angle" (I.e. heading). If you used setSpeed(50), then getSpeed will return 50. If you wrote setAngle(0), then getAngle will merely return 0, not the actual heading of the car.

If you want to get the actual speed in meters per second, you should talk to the Odometers (encoder). As they can actually measure distance (and therefore speed), you can use their getSpeed method to get "the ground speed" in meters per second. If you want to get a heading, then you will have to rely on a HeadingSensor (I.e. a gyroscope or the newly added MPU6050).

I am closing this issue, as I believe the features that are requested are already there! :)

closed 2016-01-30 by platisd

DouglasPearless, 2016-01-30:
Thanks for the comprehensive feedback!

platisd, 2016-01-30:
You're welcome! I am eager to see how you will use this library! :)

DouglasPearless, 2016-01-30:
The first thing I need to solve is the two DC drive motors run at different speeds and how I can get the car to drive in a straight line.

platisd, 2016-01-30:
Hehe, that can be a very complicated task, the complexity of which increases along with the extent/accuracy of how straight you want it to go. I have some ideas, but if you don't mind, let's continue this discussion elsewhere! You can reach me at dimitris@plat.is :)