platisd/smartcar_shield #5: Cruise control suffers from runaway due to temporary sensor failure

michalpalka, 2018-04-30:
The cruise control functionality of the library can cause runaway as a result of a temporary sensor failure. As a consequence, the controlled rover moves in the direction opposite to the requested one at maximum speed.

The problem affects the cruise control functionality enabled using `car.enableCruiseControl()`. We observed it with [this rover type](https://platis.solutions/blog/2016/06/17/turning-software-engineers-into-makers/).

The problem manifests when the rover is commanded to drive forward, but is still standing and the odometers temporarily report high moving speed due to signal transmission failure. The controller will then apply negative torque (meant for braking), which accelerates the rover backwards. As the rover accelerates backwards, the odometers correctly report increasing speed (the odometers report the absolute speed, not the direction), which is interpreted as the rover moving forward too fast. This results in more negative torque being applied, and a complete runaway with the rover racing backwards.

The following sketch can be used to demonstrate the problem.

```
#include <Smartcar.h>

Odometer encoderLeft(188), encoderRight(188);
Gyroscope gyro(22);
Car car;

const int encoderLeftPin = 2;
const int encoderRightPin = 3;
const float fSpeed = 0.3; //a ground speed (m/sec) for going forward
bool sw = false;

void setup() {
  Serial3.begin(9600);
  encoderLeft.attach(encoderLeftPin);
  encoderRight.attach(encoderRightPin);
  gyro.attach();
  delay(1500);
  gyro.begin(60);
  encoderLeft.begin();
  encoderRight.begin();
  car.begin(encoderLeft, encoderRight, gyro); //initialize the car using the encoders
  car.enableCruiseControl(); //using default PID values
  car.setAngle(0);
 // car.enableCruiseControl(12,5,17,30); //using custom values and control frequency
  car.setSpeed(fSpeed);
}

void loop() {
  car.updateMotors();
  handleInput();
}

int state = 0;

void handleInput() {
  int t = millis();
  if (t >= 1000 && state < 1) {
    car.setSpeed(fSpeed);
    state = 1;
  } else if (t >= 3000 && state < 2) {
    car.setSpeed(0);
    state = 2;
  }

}
```

The sketch uses the cruise control functionality, requesting to drive forward at 1 s after setup, and then stopping at 3 s. The correct behaviour with the sketch is the rover driving forward and then stopping.

We managed to trigger erroneous behaviour by using the rover with depleted batteries. With that setup, the rover will occasionally race backwards instead of moving forward. We observed incorrect behaviour once every 4 attempts or so, but this probability is greatly dependent on the state of the batteries. We also observed it by using an incomplete battery pack (6 1.5 V batteries instead of 8), giving the measured voltage 8.4 V dropping to 6.4 V under load (we could not measure voltage under load reliably due to how fast was changing). Connecting the Arduino board to USB power makes the problem go away completely.

We traced the problems to the odometers returning high speeds temporarily when the rover is standing or moving slowly. We don’t know what causes the incorrect readouts. We suspect that the voltage ripple from the motor controller is interpreted as signal edges from the odometers.

We believe that the cause of the incorrect readouts should be investigated, but the fundamental problem here is that the control turns a small disturbance (or input corruption) into a complete runaway. Using odometers that also report the direction of movement in addition to the speed would solve the problem. With the current odometers, the controller should be somehow constrained to avoid the runaway.

michalpalka, 2018-04-30:
@platisd @Shaun-McMurray @guslauer @ArchiCo @FFrostYY

platisd, 2018-05-02:
I will start looking into this. I think the problem that needs to be investigated is whether there are ways to safeguard against faulty odometer readings however these may occur.

As you correctly mentioned, the "simple" Odometers cannot determine direction of movement and that is why some "special" logic is in place when it comes to stopping or changing direction of movement. Specifically, when the vehicle is driving forward and we command it to start going backward, from an odometer perspective there is no difference between these two states. That's why we try to [stop](https://github.com/platisd/smartcar_shield/blob/master/src/Car.cpp#L100) the car first by applying some torque in the opposite direction

To accelerate in the investigation, it would be great to provide some logs, i.e. the speed of the car (`Car::getSpeed()`) and the current state. Since the problem is not observed when the board is connected via USB, you should use a Bluetooth module or any other wireless means of communication.

guslauer, 2018-05-05:
Reverse:
Car speed | Right odometer speed | Left odometer speed | Battery pack voltage
--- | --- | --- | ---
0.35 | 0.11 | 0.38 | 9.59
0.35 | 0.11 | 0.38 | 9.59
0.35 | 0.11 | 0.38 | 9.52
0.35 | 0.11 | 0.38 | 9.59
0.35 | 0.11 | 0.38 | 9.57
0.35 | 0.11 | 0.38 | 9.57
0.35 | 0.11 | 0.38 | 9.52
0.35 | 0.11 | 0.38 | 9.57
0.35 | 0.11 | 0.38 | 9.57
0.35 | 0.11 | 0.38 | 9.55
0.35 | 0.11 | 0.38 | 9.57
0.35 | 0.11 | 0.38 | 9.55
0.35 | 0.11 | 0.38 | 9.57
0.35 | 0.11 | 0.38 | 9.57
0.35 | 0.11 | 0.38 | 9.47
0.35 | 0.11 | 0.38 | 8.33
0.35 | 0.11 | 0.38 | 9.30
0.35 | 0.11 | 0.38 | 9.50
0.35 | 0.11 | 0.38 | 9.47
0.35 | 0.11 | 0.38 | 9.47
0.35 | 0.11 | 0.38 | 9.47
0.35 | 0.11 | 0.38 | 9.47
0.35 | 0.11 | 0.38 | 9.47
0.35 | 0.11 | 0.38 | 8.98
0.35 | 0.11 | 0.38 | 9.45
0.35 | 0.11 | 0.38 | 8.01
0.35 | 0.11 | 0.38 | 9.38
0.35 | 0.11 | 0.38 | 9.13
0.35 | 0.11 | 0.38 | 9.30
0.35 | 0.11 | 0.38 | 9.40
0.35 | 0.11 | 0.38 | 9.35
0.35 | 0.11 | 0.38 | 9.38
0.35 | 0.11 | 0.38 | 7.52
0.35 | 0.11 | 0.38 | 9.38
0.35 | 0.11 | 0.38 | 9.35
0.35 | 0.11 | 0.38 | 9.13
0.35 | 0.11 | 0.38 | 6.69
0.35 | 0.11 | 0.38 | 8.76
0.35 | 0.11 | 0.38 | 9.25
0.35 | 0.11 | 0.38 | 9.13
0.35 | 0.11 | 0.38 | 9.20
0.35 | 0.11 | 0.38 | 9.01
0.35 | 0.11 | 0.38 | 9.23
0.35 | 0.11 | 0.38 | 9.25
0.35 | 0.11 | 0.38 | 8.45
0.35 | 0.11 | 0.38 | 7.69
0.35 | 0.11 | 0.38 | 6.49
0.35 | 0.11 | 0.38 | 9.13
0.35 | 0.11 | 0.38 | 8.69
0.35 | 0.11 | 0.38 | 8.96
0.35 | 3.34 | 0.38 | 9.16
0.35 | 3.34 | 0.38 | 9.08
0.35 | 3.34 | 0.38 | 9.13
0.35 | 3.34 | 0.38 | 9.25
0.35 | 3.34 | 0.38 | 9.42
0.35 | 3.34 | 0.38 | 9.45
0.35 | 3.34 | 0.38 | 9.38
0.35 | 3.34 | 0.38 | 9.13
0.35 | 3.34 | 0.38 | 9.33
0.35 | 3.34 | 0.38 | 9.25
0.35 | 3.34 | 0.38 | 9.16
0.35 | 3.34 | 0.38 | 9.20
0.35 | 3.34 | 0.38 | 6.93
0.35 | 3.34 | 0.38 | 9.01
0.35 | 0.01 | 0.38 | 8.47
0.35 | 0.01 | 0.38 | 8.74
0.35 | 0.01 | 0.00 | 7.15
0.35 | 0.06 | 0.16 | 8.03
0.35 | 0.06 | 0.16 | 9.20
0.35 | 0.10 | 0.09 | 9.08
0.35 | 0.10 | 0.16 | 9.20
0.35 | 0.07 | 0.16 | 9.30
0.35 | 0.07 | 0.09 | 9.28
0.35 | 0.10 | 0.09 | 9.30
0.35 | 0.10 | 0.14 | 8.98
0.35 | 0.10 | 0.14 | 9.38
0.35 | 0.10 | 0.07 | 7.84
0.35 | 0.05 | 0.07 | 9.40
0.35 | 0.05 | 0.07 | 9.33
0.35 | 0.05 | 0.07 | 9.40
0.35 | 0.05 | 0.07 | 9.45
0.35 | 0.05 | 0.07 | 9.47
0.35 | 0.05 | 0.07 | 9.47
0.35 | 0.05 | 0.07 | 9.38
0.35 | 0.05 | 0.07 | 9.47
0.35 | 0.05 | 0.07 | 9.50
0.35 | 0.05 | 0.07 | 9.45
0.35 | 0.05 | 0.07 | 9.23
0.35 | 0.05 | 0.07 | 9.40
0.35 | 0.05 | 0.07 | 9.45
0.35 | 0.05 | 0.07 | 9.38
0.35 | 0.05 | 0.07 | 9.42
0.35 | 0.05 | 0.07 | 9.35
0.35 | 0.05 | 0.07 | 9.35
0.35 | 0.05 | 0.07 | 9.38
0.35 | 0.05 | 0.07 | 8.79
0.35 | 0.05 | 0.07 | 8.28
0.35 | 0.05 | 0.07 | 6.71
0.00 | 0.05 | 0.07 | 9.18
0.00 | 0.05 | 0.07 | 9.42
0.00 | 0.05 | 0.07 | 9.45
0.00 | 0.05 | 0.07 | 9.50
0.00 | 0.05 | 0.07 | 9.50
0.00 | 0.05 | 0.07 | 9.50
0.00 | 0.05 | 0.07 | 9.52
0.00 | 0.05 | 0.07 | 9.52
0.00 | 0.05 | 0.07 | 9.52

Not moving:

guslauer, 2018-05-05:
# Logs
Here are some logs from today's troubleshooting session with @Shaun-McMurray. We tried to replicate the reversing behavior with the battery pack having different voltages, and it was, as previously mentioned, very apparent that this reversing behavior happened more often when the voltage was low. The logs were sent over Bluetooth, thus no USB cable was connected to the car during the testing.  

The car speed was retrieved by `Car::getSpeed()`, odometer speed by `odometer::getSpeed()` and the 'battery pack voltage' from a voltage divider connected to the battery pack and to an analog pin on the Arduino.

In addition to the reversing behavior of the car, we also observed that the car did not move at all at times. See the section at the bottom for logs retrieved when that happened.

#### Normal behavior (control):
Car speed | Right odometer speed | Left odometer speed | Battery pack voltage
--- | --- | --- | ---
0.40 | 0.11 | 0.10 | 9.64
0.40 | 0.11 | 0.10 | 9.38
0.40 | 0.11 | 0.10 | 9.45
0.40 | 0.11 | 0.10 | 9.64
0.40 | 0.11 | 0.10 | 9.64
0.40 | 0.11 | 0.10 | 9.50
0.40 | 0.11 | 0.10 | 8.96
0.40 | 0.11 | 0.10 | 9.30
0.40 | 0.11 | 0.10 | 9.45
0.40 | 0.11 | 0.10 | 9.38
0.40 | 0.11 | 0.10 | 9.42
0.40 | 0.11 | 0.10 | 9.42
0.40 | 0.11 | 0.10 | 9.40
0.40 | 0.11 | 0.10 | 9.40
0.40 | 0.11 | 0.10 | 8.47
0.40 | 0.11 | 0.10 | 8.28
0.40 | 0.11 | 0.10 | 6.71
0.40 | 0.11 | 0.10 | 9.08
0.40 | 0.11 | 0.10 | 8.54
0.40 | 0.11 | 0.10 | 8.91
0.40 | 0.11 | 0.10 | 9.13
0.40 | 0.11 | 0.10 | 8.98
0.40 | 0.11 | 0.00 | 9.06
0.40 | 0.11 | 0.00 | 6.10
0.40 | 0.11 | 0.08 | 8.86
0.40 | 0.00 | 0.22 | 7.20
0.40 | 0.00 | 0.13 | 8.45
0.40 | 0.00 | 0.26 | 6.13
0.40 | 0.07 | 0.16 | 7.69
0.40 | 0.07 | 0.30 | 8.86
0.40 | 0.09 | 0.36 | 8.54
0.40 | 0.19 | 0.21 | 8.76
0.40 | 0.14 | 0.24 | 6.86
0.40 | 0.26 | 0.26 | 8.86
0.40 | 0.29 | 0.28 | 8.08
0.40 | 0.20 | 0.29 | 6.81
0.40 | 0.33 | 0.30 | 6.67
0.40 | 0.37 | 0.32 | 6.62
0.40 | 0.36 | 0.58 | 8.69
0.40 | 0.39 | 0.57 | 7.84
0.40 | 0.28 | 0.60 | 8.47
0.40 | 0.27 | 0.34 | 7.96
0.40 | 0.27 | 0.35 | 8.76
0.40 | 0.28 | 0.61 | 8.86
0.40 | 0.28 | 0.62 | 6.57
0.40 | 0.30 | 0.61 | 7.45
0.40 | 0.30 | 0.36 | 6.79
0.40 | 0.30 | 0.65 | 8.47
0.40 | 0.31 | 0.66 | 6.64
0.40 | 0.33 | 0.37 | 7.93
0.40 | 0.47 | 0.65 | 8.84
0.40 | 0.52 | 0.65 | 8.62
0.40 | 0.49 | 0.36 | 8.76
0.40 | 0.51 | 0.65 | 6.96
0.40 | 0.52 | 0.67 | 8.62
0.40 | 0.51 | 0.38 | 7.45
0.40 | 0.33 | 0.70 | 7.98
0.40 | 0.34 | 0.36 | 6.81
0.40 | 0.34 | 0.68 | 6.76
0.40 | 0.50 | 0.69 | 8.79
[... 891 lines cut ...]
0.40 | 0.40 | 0.41 | 9.67
0.40 | 0.40 | 0.41 | 9.67
0.40 | 0.40 | 0.41 | 9.67
0.40 | 0.40 | 0.41 | 9.67
0.40 | 0.40 | 0.41 | 9.67
0.40 | 0.40 | 0.41 | 9.45
0.40 | 0.40 | 0.41 | 9.64
0.40 | 0.40 | 0.41 | 9.67
0.40 | 0.40 | 0.41 | 9.67
0.40 | 0.40 | 0.41 | 9.67
0.40 | 0.40 | 0.41 | 9.67
0.40 | 0.40 | 0.41 | 9.69
0.40 | 0.40 | 0.41 | 9.67
0.40 | 0.40 | 0.41 | 9.64
0.40 | 0.40 | 0.41 | 9.67
0.40 | 0.40 | 0.41 | 9.30
0.00 | 0.40 | 0.41 | 9.69
0.00 | 0.40 | 0.41 | 9.67
0.00 | 0.40 | 0.41 | 9.69
0.00 | 0.40 | 0.41 | 9.69
0.00 | 0.40 | 0.41 | 9.69
0.00 | 0.40 | 0.41 | 9.69
0.00 | 0.40 | 0.41 | 9.69
0.00 | 0.40 | 0.41 | 9.69
0.00 | 0.40 | 0.41 | 9.69

platisd, 2018-05-05:
Is it when we see the peak in the odometer speed that the car starts driving the wrong way or has it already started reversing when these unusually high speeds are logged?

In other words can we say for a fact that these high speed readings are causing the weird behavior?

guslauer, 2018-05-05:
The reversing is triggered when a peak in the odometer speed occurs. So yes, it seems that this is the direct cause of the reversing behavior.

platisd, 2018-05-05:
Thanks a lot for your feedback. It's hard for me to reproduce this so these logs should settle this odometer malfunction as the cause of the defect.

Figuring out why the odometers behave this why would be rather hard as they are cheap pieces of hardware that shouldn't be taken very seriously. So I think whatever fix should be there and not in the `Car` logic. We should somehow ensure that the readings are as trustworthy as possible.

Can you try to increase [this value](https://github.com/platisd/smartcar_shield/blob/master/src/Odometer.cpp#L24) to `1000` or more so to see if the problem gets mitigated?

platisd, 2018-05-19:
The problem can be easily reproduced on any car (regardless of how good or bad the odometers are) by using the following simple code and process:

````cpp
/* Drive using cruise control for a specified amount of time and then stop */
#include <Smartcar.h>

Odometer encoderLeft(94), encoderRight(94);
Car car;
const unsigned long TIME_TO_STOP = 10000; // In milliseconds
float carSpeed = 0.3;

void setup() {
  encoderLeft.attach(2);
  encoderRight.attach(3);
  encoderLeft.begin();
  encoderRight.begin();
  car.begin(encoderLeft, encoderRight);
  car.enableCruiseControl();
  car.setSpeed(carSpeed);
}

void loop() {
  car.updateMotors();
  if (millis() > TIME_TO_STOP) {
    car.stop();
  }
}
```
Under normal circumstances, the above sketch makes the car move for 15 seconds (or whatever you set the `TIME_TO_STOP` variable to) and then stops. Faulty readings can be simulated by pushing the car forward while making sure that the wheels are touching the ground and thus spinning faster than they usually do. This could also be an actual scenario by the way, if the car is on a downhill and moves faster than it should.
In that case, the car moves towards the wrong direction. I am trying to figure out what is the best way of doing this without affecting too much the more common use cases. :+1:

platisd, 2018-05-19:
@michalpalka @Shaun-McMurray @guslauer @ArchiCo @FFrostYY feel free to try the fix found at #9 out or review the code. It should be fixing the problem.

If no objections I will merge it in a couple of days.

commit c45224dd (2018-05-19): Fix runaway issues in cruise control

When the measured speed was increased from outside factors
during cruise control, the car while trying to slow down,
would eventually start accelerating towards the opposite
direction. This problem surfaced due to incorrect Odometer
readings, however could happen in other (valid) occasions.
The main problem behind it is that (most) Odometers cannot
provide the direction of movement.

Fixes #5
PR #9 (2018-05-21): Fix runaway issues in cruise control
When the measured speed was increased from outside factors during cruise control, the car while trying to slow down, would eventually start accelerating towards the opposite direction. This problem surfaced due to incorrect Odometer readings, however could happen in other (valid) occasions.
The main cause behind this problem is that (most) Odometers cannot provide the direction of movement.

Fixes #5
closed 2018-05-21 by platisd