platisd/smartcar_shield #60: support servo drivemotor?

samuk, 2022-08-22:
## Description

## Definition of Done

platisd, 2022-08-22:
At first I thought you meant an ESC ([which is supported](https://platisd.github.io/smartcar_shield/class_servo_motor.html)) but I now see that you mean those continuous rotation "servo" motors.

Can you remind me how do you set the speed for them? It shouldn't be hard to implement something if not already supported.

samuk, 2022-08-22:
it's just PWM https://dronebotworkshop.com/esp32-servo/

I think you already have the esp32servo lib installed?

platisd, 2022-08-22:
The particular sensors are a little special in the sense that you cannot set a "speed" on them. They just move at a set speed forward or backward. But anyway, if you replace `BrushedMotor` with `ServoMotor` in [SimpleCar.ino](https://github.com/platisd/smartcar_shield/blob/master/examples/Car/SimpleCar/SimpleCar.ino) it should work.

samuk, 2022-08-22:
Great I'll give it a go, thanks

closed 2022-08-22 by samuk