platisd/smartcar_shield #61: supported IMU/ motor driver  list

samuk, 2022-08-22:
## Description

Is there a list of supported IMU anywhere? Or is it just the GY50?

I'd be particularly interested in any of the breakout garden IMU? 
https://shop.pimoroni.com/collections/breakout-garden?q=robot 

## Definition of Done
Documentation includes list of supported hardware,  or clarifies rhat it is only GY50 that is supported

platisd, 2022-08-22:
Currently only the GY-50 is supported as mentioned in [Supported sensors](https://github.com/platisd/smartcar_shield#supported-sensors)

samuk, 2022-08-22:
Thanks. I might see if I can add support for https://learn.adafruit.com/adafruit-bno055-absolute-orientation-sensor/arduino-code

closed 2022-08-22 by samuk