platisd/smartcar_shield #62: DRV8830

samuk, 2022-08-24:
## Description

Would you be interested in supporting https://github.com/adafruit/Adafruit_DRV8830 under your library?

## Definition of Done

platisd, 2022-08-24:
Sure, why not? The placeholder I2cMotor class was created with something like that in mind.

samuk, 2022-08-25:
Great! This is the one I'd use https://shop.pimoroni.com/products/drv8830-dc-motor-driver-breakout?variant=30093450739795 there are a few available

platisd, 2022-08-25:
I suggest the following steps forward since I don't have this piece of HW:
1. Get your hands on the hardware
2. Make it work with some library/code
3. Send me a link to that library/code (please do not use software that's licensed under GPL) along with exactly what address were used etc.
4. I integrate it to this library

Does that sound good?

samuk, 2022-08-25:
Great!

closed 2023-01-23 by samuk