platisd/smartcar_shield #63: BNO055?

samuk, 2022-09-04:
## Description

Add support for https://github.com/adafruit/Adafruit_BNO055

Various open hardware boards are available:
https://www.smart-prototyping.com/Zio-9DOF-IMU-BNO055.html
https://www.adafruit.com/product/4646

If I post you one would you have a go at getting it working?

## Definition of Done

platisd, 2022-09-04:
I'm afraid I can't promise I will have the time for this in the near future so my suggestion is to give it a go yourself first.
Specifically, if you make it adhere to the [HeadingSensor](https://github.com/platisd/smartcar_shield/blob/master/src/sensors/heading/HeadingSensor.hpp) 
interface then it's very easy adding it to the rest of the stack.
Looking at this [example](https://github.com/adafruit/Adafruit_BNO055/blob/master/examples/position/position.ino) the `z` is what I believe you should need to expose as "the heading", assuming some reasonable sensor placement on the vehicle.

I will be very happy to assist where I can. Don't know if this will be supported as a standard part of the Smartcar library, but it could definitely be an add-on type of thing.

samuk, 2022-09-06:
OK thanks

closed 2022-09-06 by samuk