platisd/smartcar_shield #65: Web interface?

samuk, 2022-09-08:
## Description

Would you be interested in adding a web interface like this: https://github.com/701lab/ESP32-robot-car-controller#esp32-robot-car-controller

To your library? It seems like it should be possible: https://github.com/701lab/ESP32-robot-car-controller/issues/2#issuecomment-1237977819

## Definition of Done

platisd, 2022-09-09:
It surely is possible but I don't see how this is some functionality that should be included in this library. :)
`smartcar_shield` is all about abstracting the control of different common mechanical and digital components on a miniature car, not include the entire stack necessary for communication, UI etc.
Additionally, including something like this would "tie" the library to ESP32 or at least WiFi-capable microcontrollers, which is not a direction we should go.

samuk, 2022-09-09:
I thought you might say that :) I'll look into doing this as a separate repo. 

Thinking about it a bit more it would be interesting to be able to access more of the API than simple steering control. Something more like https://github.com/ayushsharma82/WebSerial

closed 2022-09-09 by samuk

platisd, 2022-09-09:
Take a look at this example of [Smartcar controls and sensors being exposed via a REST API](https://github.com/DIT112-V20/kotlin-app-arduino-sketch-ci/blob/master/arduino/SmartcarWebserver/SmartcarWebserver.ino).